MiRoboticsLab/cyberdog_ros2 #10: Failed to compile a simple example

witchoco, 2021-09-18:
download the example of minimal subscriber from ROS2 project, run colcon build, and got below errors:
mi@lubuntu:/home/jason/work/ros_apps$ colcon build --packages-up-to examples_rclcpp_minimal_subscriber
Starting >>> examples_rclcpp_minimal_subscriber
--- stderr: examples_rclcpp_minimal_subscriber                               
/home/jason/work/ros_apps/src/examples_rclcpp_minimal_subscriber/member_function_with_type_adapter.cpp:19:10: fatal error: rclcpp/type_adapter.hpp: No such file or directory
 #include "rclcpp/type_adapter.hpp"
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/subscriber_member_function_with_type_adapter.dir/member_function_with_type_adapter.cpp.o] Error 1
make[1]: *** [CMakeFiles/subscriber_member_function_with_type_adapter.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/jason/work/ros_apps/src/examples_rclcpp_minimal_subscriber/member_function_with_unique_network_flow_endpoints.cpp:92:31: error: ‘NetworkFlowEndpoint’ is not a member of ‘rclcpp’
     const std::vector<rclcpp::NetworkFlowEndpoint> & network_flow_endpoints) const
                               ^~~~~~~~~~~~~~~~~~~
/home/jason/work/ros_apps/src/examples_rclcpp_minimal_subscriber/member_function_with_unique_network_flow_endpoints.cpp:92:31: error: ‘NetworkFlowEndpoint’ is not a member of ‘rclcpp’
/home/jason/work/ros_apps/src/examples_rclcpp_minimal_subscriber/member_function_with_unique_network_flow_endpoints.cpp:92:50: error: template argument 1 is invalid
     const std::vector<rclcpp::NetworkFlowEndpoint> & network_flow_endpoints) const
                                                  ^
/home/jason/work/ros_apps/src/examples_rclcpp_minimal_subscriber/member_function_with_unique_network_flow_endpoints.cpp:92:50: error: template argument 2 is invalid
/home/jason/work/ros_apps/src/examples_rclcpp_minimal_subscriber/member_function_with_unique_network_flow_endpoints.cpp: In constructor ‘MinimalSubscriberWithUniqueNetworkFlowEndpoints::MinimalSubscriberWithUniqueNetworkFlowEndpoints()’:
/home/jason/work/ros_apps/src/examples_rclcpp_minimal_subscriber/member_function_with_unique_network_flow_endpoints.cpp:39:17: error: ‘struct rclcpp::SubscriptionOptionsWithAllocator<std::allocator<void> >’ has no member named ‘require_unique_network_flow_endpoints’
       options_1.require_unique_network_flow_endpoints =
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jason/work/ros_apps/src/examples_rclcpp_minimal_subscriber/member_function_with_unique_network_flow_endpoints.cpp:40:9: error: ‘RMW_UNIQUE_NETWORK_FLOW_ENDPOINTS_STRICTLY_REQUIRED’ was not declared in this scope
         RMW_UNIQUE_NETWORK_FLOW_ENDPOINTS_STRICTLY_REQUIRED;

.......

witchoco, 2021-09-18:
It seems there is something wrong with the environment configuration of development. Would someone give some suggestion, please~~~?

homalozoa, 2021-09-18:
Hi, @witchoco. It seems the version of ROS 2 that you are using is Rolling Ridley, but what is running on CyberDog is Foxy Fitzroy.  Hince there is no C++ header named `rclcpp/type_adapter.hpp`. Please check it. Thanks.

closed 2021-09-24 by homalozoa