MiRoboticsLab/cyberdog_ros2 #17: Cannot Log in to the Cyberdog

Entromorgan, 2021-09-23:
Hello, I've tried to log in to the Cyberdog onboard system with the following steps but failed:

1. Press the start button of the Cyberdog until both lights on the head and tail shine (blue and red separately).

2. Connect the download cable (not the charging cable) with the Cyberdog (on the download port) and a PC/laptop

3. Use ```ssh mi@192.168.55.1``` (on a Ubuntu 18.04 PC) or Mobaxterm/putty (on a Windows10 laptop) to log in.

Both cases returned ```Connection timed out```.

BTW, when connect to the Windows10 laptop, it reports that the APX driver is missing.

What's the problem?

homalozoa, 2021-09-23:
Hi@Entromorgan. 

1. Can you control it with your Mobile phone after power on? If not, perhaps you have to flash a new image to recover it.
2. If so, did you try connecting to the robot with another cable? Download cable is designed for flashing.

Hope this helps.

Entromorgan, 2021-09-24:
Thanks for the timely reply.

-  I can control it with my MI 11 after power on.

- I've tried another cable but get the same results, i.e., connection time out.

Then I was wondering if this has something to do with the mode so I (1) reboot the Cyberdog (2) touch its head for 3 seconds until the headlight turns to flashed orange. 

As a result, I can log in to the system via either the download cable or the other arbitrary charging cable on both the aforementioned Windows 10 laptop and the Ubuntu 18.04 PC.

Hope it helps.

closed 2021-09-24 by Entromorgan