MiRoboticsLab/cyberdog_ros2 #19: Could you provide the urdf file for CyberDog?

ZiwenZhuang, 2021-09-26:
## Brief modification description
providing URDF file for cyberdog

## Detailed information
We wish that by having the urdf file, we can make a simulation interface based on PyBullet. And robot learning research based on cyberdog can be performed

zhangsining734, 2021-09-27:
附议

homalozoa, 2021-09-27:
再等等吧，领导不让给。我也没办法...

zbwu, 2021-09-29:
这个文件对开发很关键，当然也涉及很多产品商业信息。

借这个地方问一下，驱动控制代码和URDF是不是在Cyberdog的开源计划里面？

homalozoa, 2021-09-29:
> 这个文件对开发很关键，当然也涉及很多产品商业信息。
> 
> 借这个地方问一下，驱动控制代码和URDF是不是在Cyberdog的开源计划里面？
> 

@zbwu 在计划里，只不过还需要等等。

tpoisonooo, 2021-11-18:
> 再等等吧，领导不让给。我也没办法...

缺主控逻辑好比缺灵魂，硬件不本质啊。不赶紧交给 contributor 瞎折腾，死守这那玩意儿有啥用。

a52521255, 2021-11-18:
附议附议！不然能做的很有限啊

homalozoa, 2021-11-18:
> > 这个文件对开发很关键，当然也涉及很多产品商业信息。
> > 借这个地方问一下，驱动控制代码和URDF是不是在Cyberdog的开源计划里面？
> 
> @zbwu 在计划里，只不过还需要等等。

最近又问了下领导，大概率不会开源了……离谱

MarkFzp, 2021-12-02:
没有urdf就没法做放到仿真环境里，很多科研方向就没法进行，宇树就开源了urdf和运动控制的sdk

MarkFzp, 2021-12-02:
> > > 这个文件对开发很关键，当然也涉及很多产品商业信息。
> > > 借这个地方问一下，驱动控制代码和URDF是不是在Cyberdog的开源计划里面？
> > 
> > 
> > @zbwu 在计划里，只不过还需要等等。
> 
> 最近又问了下领导，大概率不会开源了……离谱

没有urdf就没法做放到仿真环境里，很多科研方向就没法进行，宇树就开源了urdf和运动控制的sdk

ZiwenZhuang, 2021-12-02:
有人有高精度扫描仪嘛hhhhh，干脆拆了自己扫描一个，然后标定。。。。

closed 2021-12-25 by homalozoa

MAVProxyUser, 2022-09-05:
URDF was ever provided?

Yaoying1, 2023-03-29:
urdf文件还会开源吗

MAVProxyUser, 2023-03-29:
The Cyberdog team is dead. All good employees, quit, or do not like their job. I hear the manager of team is awful and does not care about open source @Yaoying1

Yaoying1, 2023-03-29:
> 

我去一个机器人描述文件而已 没必要死守着吧 这还当个宝了

MAVProxyUser, 2023-03-29:
Email project manager and complain he does not care about end users that purchased his project. [xiangdiyun@xiaomi.com](mailto:xiangdiyun@xiaomi.com)

Yaoying1, 2023-03-29:
> 给项目经理发电子邮件，抱怨他不关心购买他项目的最终用户。[xiangdiyun@xiaomi.com](mailto:xiangdiyun@xiaomi.com)

你好 有相关社区或者群来进行讨论、交流吗