MiRoboticsLab/cyberdog_ros2 #21: Dependency problem of building ros2(mini.reps) from source

zbwu, 2021-09-29:
#rmw_implementation: No definition of [rmw_connextdds] for OS version [bionic]

Missing rmw_connextdds in mini.repos

#imu_tools: Cannot locate rosdep definition for [rviz_imu_plugin]

imu_tools depends on rviz_imu_plugin that not include in mini.repos

homalozoa, 2021-09-29:
imu_tools里的rviz插件在mini.repos的仓里已经被删掉了（是我fork的版本，不过这个工具之后我会删掉，项目早期使用过），此外，我们目前不支持rti的中间件，也就是rmw_connextdds，如果对编译造成影响，我可以提PR把这部分相关的内容删除掉。我们推荐使用cyclonedds和fastdds。

The rviz plugin of imu_tools was removed in the repository listed in the mini.repos.(I forked it, and I'll remove it later). We do not support DDS middle ware provided by rti, aka rmw_connextdds. If it made troubles, I will submit a pr to remove it. We recommend you run ROS 2 with cyclonedds or fastdds.

zbwu, 2021-09-29:
RTI的其他几个包都在，就缺少一个，所以提示依赖问题，可以考虑都删除掉，这个应该用不上。

imu_tools的代码库里面好像还是包括rviz的插件

homalozoa, 2021-09-29:
提个PR删了吧

closed 2021-10-17 by homalozoa