MiRoboticsLab/cyberdog_ros2 #43: 刷机导致蓝牙连接不上

liseri, 2021-11-08:
硬件平台：铁蛋

问题：刷机后 蓝牙连接失败；之前刷过一次小米镜像，没出问题，我今天刷了一次Jetpack官方镜像，又刷回小米镜像，发现手机app连不上蓝牙了， 网上的两个版本镜像都试了，都不行；我发现有bluetooth有报错，如下；

报错详情：bluetooth_ros2.service报错如下：
Nov 08 02:53:57 lubuntu ros2[17548]: 2021-11-08 02:53:57,015 - gattserver.py[line:1041] - DEBUG: main: set bleStatus BLE_STATUS_NONE -> BLE_STATUS_ADV
Nov 08 02:53:57 lubuntu ros2[17548]: 2021-11-08 02:53:57,016 - gattserver.py[line:950] - INFO: RegADV: RegisterAdvertisement
Nov 08 02:53:57 lubuntu ros2[17548]: 2021-11-08 02:53:57,017 - gattserver.py[line:962] - INFO: RegATT: RegisterApplication
Nov 08 02:53:57 lubuntu ros2[17548]: 2021-11-08 02:53:57,017 - gattserver.py[line:253] - INFO: get_path: <bluetooth.gattserver.RobotApplication at / at 0x7f9ae12518>
Nov 08 02:53:57 lubuntu ros2[17548]: 2021-11-08 02:53:57,020 - gattserver.py[line:1045] - DEBUG: main init and send audio cmd BOOT_COMPLETE
Nov 08 02:53:57 lubuntu ros2[17548]: 2021-11-08 02:53:57,024 - gattserver.py[line:1048] - DEBUG: RegADV: BT enable adv, set LED_BT_ADV
Nov 08 02:53:57 lubuntu ros2[17548]: 2021-11-08 02:53:57,040 - gattserver.py[line:783] - INFO: register_app_cb: GATT application registered!!!
Nov 08 02:53:57 lubuntu ros2[17548]: 2021-11-08 02:53:57,041 - gattserver.py[line:198] - INFO: GetAll
Nov 08 02:53:57 lubuntu ros2[17548]: 2021-11-08 02:53:57,041 - gattserver.py[line:201] - INFO: returning props
Nov 08 02:53:57 lubuntu ros2[17548]: *ERROR:dbus.service:Unable to append ({'Type': 'peripheral', 'ServiceUUIDs': dbus.Array(['C420'], signature=dbus.Signature('s')), 'ManufacturerData': dbus.Dictionary({65535: dbus.Array([0, 1, 2, 3, 4], signature=dbus.Signature('y'))}, signature=dbus.Signature('qv')), 'ServiceData': dbus.Dictionary({'9999': dbus.Array([0, 1, 2, 3, 4], signature=dbus.Signature('y'))}, signature=dbus.Signature('sv')), 'LocalName': dbus.String('铁蛋ne QED_ROCE_MODIFY\nne QED_ROCE_MODIFY\nne QED_ROCE_MODIFY\nne QED_ROCE_MODIFY\n_MDS_OP_LOOKUPSNAP\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\nne _NET_PSNAP_H\nne HFI1_CAP_EXTEND\nne PCI_EXP_SLTCAP\nef void (APIENTRYP\nef Bool ( PFNGLXQ\nef void (GLAPIENTR\nFUN_EXPORT PFNGLEV\nest.Adobe-DPS-Exte\nest.Adobe-DPS-Exte\n'), 'IncludeTxPower': dbus.Boolean(True)},) to message with signature a{sv}: <class 'ValueError'>: embedded null byte
Nov 08 02:53:57 lubuntu ros2[17548]: 2021-11-08 02:53:57,048 - gattserver.py[line:794] - INFO: register_ad_cb: Advertisement registered!!!
Nov 08 02:53:57 lubuntu ros2[17548]: INFO:/var/log/bluetooth.log:register_ad_cb: Advertisement registered!!!

liseri, 2021-11-08:
更新：
更新：
更新：
我触摸头部，叮呤响一下，然后再看bluetooth_ros2.service，以上报错没有了，以上报错没有了， 但是仍然联接不上，但是仍然联接不上， bluetooth_ros2.service日志如下：
Nov 08 23:21:36 lubuntu ros2[12692]: 2021-11-08 23:21:36,336 - remote.py[line:855] - INFO: stopAutoConnect: end
Nov 08 23:21:36 lubuntu ros2[12692]: INFO:/var/log/bluetooth.log:stopAutoConnect: end
Nov 08 23:21:36 lubuntu ros2[12692]: 2021-11-08 23:21:36,337 - remote.py[line:762] - INFO: checkAndAutoDisconnect: get r
Nov 08 23:21:36 lubuntu ros2[12692]: INFO:/var/log/bluetooth.log:checkAndAutoDisconnect: get remoteStatus is 10
Nov 08 23:21:36 lubuntu ros2[12692]: 2021-11-08 23:21:36,338 - remote.py[line:764] - DEBUG: checkAndAutoDisconnect: alre
Nov 08 23:21:36 lubuntu ros2[12692]: DEBUG:/var/log/bluetooth.log:checkAndAutoDisconnect: already disconnected
Nov 08 23:21:36 lubuntu ros2[12692]: 2021-11-08 23:21:36,338 - gattserver.py[line:910] - DEBUG: current bleStatus : 1
Nov 08 23:21:36 lubuntu ros2[12692]: DEBUG:/var/log/bluetooth.log:current bleStatus : 1
Nov 08 23:21:36 lubuntu ros2[12692]: 2021-11-08 23:21:36,341 - gattserver.py[line:943] - DEBUG: send audio cmd TOUCH_ADV
Nov 08 23:21:36 lubuntu ros2[12692]: DEBUG:/var/log/bluetooth.log:send audio cmd TOUCH_ADV_OR_WIFI_CONNECT_SUCCESS from
lines 1-19/19 (END)

homalozoa, 2021-11-09:
触摸头部铁蛋会响这个过程和蓝牙正常工作没有直接联系。

整条链路是

> 触摸传感器检测到特定的触摸方式--> 唤醒蓝牙 --> 播放音乐

所以蓝牙可能还是不正常的。不过不要尝试刷NV提供的镜像到铁蛋，很多硬件环境都不一致，所以外设都是不可用的。

homalozoa, 2021-11-09:
我这边测试一下你说的流程，晚点再做沟通。

liseri, 2021-11-09:
> 我这边测试一下你说的流程，晚点再做沟通。

好的，需要我做什么测试，查看什么日志，我这都可配合

Qiupetter, 2021-11-09:
你好，请查看邮件。

Qiupetter, 2021-11-09:
> > 我这边测试一下你说的流程，晚点再做沟通。
> 
> 好的，需要我做什么测试，查看什么日志，我这都可配合

你好请查看邮件。

liseri, 2021-11-10:
已解决，是由于刷机导致丢失XXX码，导致蓝牙连接失败

closed 2021-11-10 by liseri

mailonghua, 2021-11-21:
小米的镜像的地址能分享一下吗

liseri, 2021-11-21:
> 小米的镜像的地址能分享一下吗
https://github.com/MiRoboticsLab/cyberdog_ros2/discussions/18