MiRoboticsLab/cyberdog_ros2 #54: compile err :colcon build --merge-install --packages-up-to cyberdog_bringup

dbdxnuliba, 2021-11-17:
tmi@tmi-NUC8i7BEH:~/ws_cyberdog$ colcon build --merge-install --packages-up-to cyberdog_bringup

Starting >>> cascade_lifecycle_msgs
Starting >>> ception_msgs                             
Starting >>> interaction_msgs
Starting >>> lcm_vendor
Starting >>> motion_msgs
Starting >>> behaviortreecppv3_vendor
Starting >>> grpc_vendor
Starting >>> toml11_vendor
--- stderr: ception_msgs                                                                                                                                                                         
CMake Error at /opt/ros/foxy/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:60 (message):
  execute_process(/usr/bin/python3 -m rosidl_adapter --package-name
  ception_msgs --arguments-file
  /home/tmi/ws_cyberdog/build/ception_msgs/rosidl_adapter__arguments__ception_msgs.json
  --output-dir
  /home/tmi/ws_cyberdog/build/ception_msgs/rosidl_adapter/ception_msgs
  --output-file
  /home/tmi/ws_cyberdog/build/ception_msgs/rosidl_adapter/ception_msgs.idls)
  returned error code 1:

  Traceback (most recent call last):

    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/parser.py", line 738, in parse_primitive_value_string
      value = int(value_string)

  ValueError: invalid literal for int() with base 10: '0xFFFFFFFFFFFFFFFF'

  

  During handling of the above exception, another exception occurred:

  

  Traceback (most recent call last):

    File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
      return _run_code(code, main_globals, None,
    File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
      exec(code, run_globals)
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/__main__.py", line 19, in <module>
      sys.exit(main())
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/main.py", line 53, in main
      abs_idl_file = convert_to_idl(
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/__init__.py", line 24, in convert_to_idl
      return convert_srv_to_idl(
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/srv/__init__.py", line 28, in convert_srv_to_idl
      srv = parse_service_string(package_name, input_file.stem, content)
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/parser.py", line 842, in parse_service_string
      request_message = parse_message_string(
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/parser.py", line 528, in parse_message_string
      constants.append(Constant(type_string, name, value))
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/parser.py", line 326, in __init__
      self.value = parse_primitive_value_string(
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/parser.py", line 740, in parse_primitive_value_string
      raise ex

homalozoa, 2021-11-17:
Hi. Could you provide more information about your platform?

closed 2021-11-22 by homalozoa

dbdxnuliba, 2021-12-05:
kinetic ubuntu16.04