MiRoboticsLab/cyberdog_ros2 #55: compier error

Peterxjerk, 2021-11-17:
Please provide information for your bug report:

- if you're using from source, provide the Git commit hash
  (via `git log -1 --format="%H" | xargs`):  c8f2ce285115f0d55b5e0dfafd435849cb06becc

- provide all information that is needed to analyze and reproduce the bug
>mi@lubuntu:~/ros_apps$ colcon build --merge-install --install-base $OUTPUT_DIR
Starting >>> cascade_lifecycle_msgs
Starting >>> ception_msgs
Starting >>> interaction_msgs                                                           
Starting >>> motion_msgs
Starting >>> toml11_vendor                                                                                                             
Starting >>> lcm_vendor
Finished <<< toml11_vendor [2.93s]                                                                                                                                             
Starting >>> behaviortreecppv3_vendor
Finished <<< lcm_vendor [3.15s]                                                                                                                        
Starting >>> lcm_translate_msgs                                                                                                                     
Finished <<< lcm_translate_msgs [2.04s]                                                                                                                  
Starting >>> ception_base
Finished <<< cascade_lifecycle_msgs [6.64s]                                                                                                                
Starting >>> cyberdog_utils
Finished <<< ception_base [2.25s]                                                                                                              
Starting >>> cyberdog_ception_bridge
Finished <<< cyberdog_ception_bridge [2.23s]                                                                                                      
Starting >>> mpg123_vendor
Finished <<< ception_msgs [11.5s]                                                                                                                     
Starting >>> grpc_vendor
Finished <<< cyberdog_utils [4.85s]
Starting >>> cyberdog_motion_bridge
[17.039s] WARNING:colcon.colcon_cmake.task.cmake.build:Could not run installation step for package 'grpc_vendor' because it has no 'install' target             
Finished <<< grpc_vendor [1.43s]                                                                                                                                
Starting >>> xiaoai_sdk_vendor
Finished <<< motion_msgs [13.2s]                                                                                                                                  
Finished <<< mpg123_vendor [3.51s]                                                                                                                                 
Starting >>> automation_msgs
Starting >>> sdl2_vendor                                                                                                                                               
Finished <<< cyberdog_motion_bridge [2.29s]                                                                                                                            
Starting >>> bcmgps_plugin
[18.965s] WARNING:colcon.colcon_cmake.task.cmake.build:Could not run installation step for package 'xiaoai_sdk_vendor' because it has no 'install' target        
Finished <<< xiaoai_sdk_vendor [1.73s]                                                                                                                           
Starting >>> cyberdog_decisionutils
Finished <<< interaction_msgs [15.3s]                                                                                                                              
Starting >>> live_stream
Finished <<< behaviortreecppv3_vendor [12.5s]                                                                                                          
Starting >>> cyberdog_bt_engine
Finished <<< bcmgps_plugin [2.32s]                                                                                                         
Starting >>> cyberdog_bms
Finished <<< cyberdog_decisionutils [2.78s]                                                                                                                                    
Starting >>> bluetooth
Finished <<< live_stream [2.59s]                                                                                                                                               
Starting >>> wifirssi
Finished <<< cyberdog_bt_engine [4.11s]                                                                                                           
Starting >>> cyberdog_body_state                                                                                                                                             
Finished <<< cyberdog_bms [5.28s]                                                                                                                              
Starting >>> cyberdog_camera
Finished <<< bluetooth [4.13s]                                                                                                                                             
Starting >>> cyberdog_gps
Finished <<< automation_msgs [9.40s]                                                                                                                              
Starting >>> cyberdog_interfaces                                                                                                                           
Finished <<< wifirssi [5.34s]                                                                                                                              
Starting >>> cyberdog_lightsensor
Finished <<< cyberdog_interfaces [1.59s]                                                                                                                              
Starting >>> cyberdog_decisionmaker
--- stderr: cyberdog_body_state                                                                                                                                       
CMake Error at CMakeLists.txt:22 (find_package):
  By not providing "Findcyberdog_utils.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "cyberdog_utils", but CMake did not find one.

  Could not find a package configuration file provided by "cyberdog_utils"
  with any of the following names:

    cyberdog_utilsConfig.cmake
    cyberdog_utils-config.cmake

  Add the installation prefix of "cyberdog_utils" to CMAKE_PREFIX_PATH or set
  "cyberdog_utils_DIR" to a directory containing one of the above files.  If
  "cyberdog_utils" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< cyberdog_body_state [5.92s, exited with code 1]
Aborted  <<< cyberdog_decisionmaker [2.52s]                                                                                                   
Aborted  <<< cyberdog_camera [6.50s]                                                                                                                                          
Aborted  <<< cyberdog_lightsensor [4.85s]                                                                                                      
Aborted  <<< cyberdog_gps [11.6s]                                                                           
Aborted  <<< sdl2_vendor [4min 30s]                                           

Summary: 24 packages finished [4min 48s]
  1 package failed: cyberdog_body_state
  5 packages aborted: cyberdog_camera cyberdog_decisionmaker cyberdog_gps cyberdog_lightsensor sdl2_vendor
  5 packages had stderr output: cyberdog_body_state cyberdog_camera cyberdog_gps cyberdog_lightsensor sdl2_vendor
  13 packages not processed

so how can I recorrect this error? thank you

homalozoa, 2021-11-22:
应该是与本机的环境冲突了，稍后我更新一个部署教程吧。

homalozoa, 2021-11-22:
以下是我的环境编译的log，环境为Arch Linux， x86-64。因为整段log太长，所以我只截取了最后的一段。

`                   
Summary: 43 packages finished [1min 39s]
  9 packages had stderr output: audio_assistant behaviortreecppv3_vendor cyberdog_audio cyberdog_bms cyberdog_camera grpc_vendor lcm_vendor live_stream sdl2_vendor
`

closed 2021-12-21 by homalozoa