MiRoboticsLab/cyberdog_ros2 #59: 铁蛋语音唤醒后怎么触发自己得语音识别程序

wgqabc, 2021-11-19:
叫铁蛋铁蛋后使用自己得语音识别程序，应该怎么做？

homalozoa, 2021-11-22:
建议尝试移植其他支持框架（非小米小爱），目前小爱同学并不支持此类应用

northdk, 2022-01-22:
Hi，wgqabc！
我们可以给你提供一个思路：
1. 在铁蛋里移植自己的语音识别程序；
2. 这里是唤醒入口，添加标识位： [wakeup](https://github.com/MiRoboticsLab/cyberdog_ros2/blob/main/cyberdog_interaction/cyberdog_audio/audio_assistant/src/ai_keyword.cpp)
函数： ` static void vpm_wakeup_data_handle(
  const void *, const vpm_callback_data_t * data, int)`
3. 这里是识别入口，把数据指向你自己的唤醒程序：[work loop](https://github.com/MiRoboticsLab/cyberdog_ros2/blob/main/cyberdog_interaction/cyberdog_audio/audio_assistant/src/ai_keyword.cpp) 
函数：  `void ai_vpm_work_loop()`
这是你要处理的代码片段：
   `vpm_input_buf_t vpm_data; 
    vpm_msg_queue.DeQueue(vpm_data); 
    input.raw = reinterpret_cast<void *>(vpm_data.start); 
    input.size = vpm_data.length; 
    vpm_process(&input);`
其中：input是录音数据，vpm_process替换成你自己的识别函数。

希望能帮到你。

ythdwl, 2022-01-27:
这个可以认真学习了

closed 2022-02-09 by mi-CyberDog