MiRoboticsLab/cyberdog_ros2 #64: 运动控制相关的开发

xygh, 2021-11-21:
## Brief modification description
想用铁蛋做些运动控制相关的开发。
## Detailed information
铁蛋的硬件主要包括：电源板、运动控制板（MR813+两块STM32，相当于mini cheetah上的up board和SPine），英伟达主控板和扩展板，12个电机驱动器。目前看开源的部分主要是英伟达主板上跑的代码。这部分主要能做的开发是感知类的开发，例如视觉、语音等，这部分的开源和其它ROS教育类的机器人没有什么区别，体现不出四足的特点。运动控制板上有MR813和两块STM32的debug口，但在没有硬件IO分配信息的情况下，改代码也显得难度巨大。不知道官方有没有底层运动控制开源的计划？另外想了解一下现在电机驱动CAN总线上的协议和mini cheetah上的是不是一样的？谢谢！

a52521255, 2021-11-21:
最后一个问题同求，这样最不济可以自己用upboard（这套nx是否可以利用，看io支持两路spi）和spine把狗子驱动起来

xygh, 2021-11-21:
如果不行就只能把驱动器也换了，就是拆起来有点费劲，其它的还好。

a52521255, 2021-11-22:
> 如果不行就只能把驱动器也换了，就是拆起来有点费劲，其它的还好。

问题是都换了，那铁蛋就用不上啥了
[image]

xygh, 2021-11-22:
> > 如果不行就只能把驱动器也换了，就是拆起来有点费劲，其它的还好。
> 
> 问题是都换了，那铁蛋就用不上啥了 [image]

xygh, 2021-11-22:
坐等官方的回复

homalozoa, 2021-11-22:
我已经尽力推动运动控制开源了hhhh，领导给的方案是搞个类似宇树SDK一样的东西开出来。也许下个月或者明年年初吧

xygh, 2021-11-23:
> 我已经尽力推动运动控制开源了hhhh，领导给的方案是搞个类似宇树SDK一样的东西开出来。也许下个月或者明年年初吧

了解了，感谢回复

szg189, 2022-01-13:
同求小米铁蛋电机的CAN协议，跪求大佬们

mi-CyberDog, 2022-01-19:
电机控制相关的LCM控制请参考https://github.com/MiRoboticsLab/cyberdog_motor_sd

closed 2022-01-19 by mi-CyberDog

MAVProxyUser, 2022-09-05:
Something new here regarding CAN
https://github.com/bgkatz/motorcontrol/commit/4c443e6aa1341babeb4229dadb221d366d2ce639

MAVProxyUser, 2022-09-05:
Se also: https://github.com/zbwu/athena_adapter/blob/main/software/MotorCtrl/protocol.py#L89

18cmbanana, 2023-01-31:
> Something new here regarding CAN [bgkatz/motorcontrol@4c443e6](https://github.com/bgkatz/motorcontrol/commit/4c443e6aa1341babeb4229dadb221d366d2ce639)

> Something new here regarding CAN [bgkatz/motorcontrol@4c443e6](https://github.com/bgkatz/motorcontrol/commit/4c443e6aa1341babeb4229dadb221d366d2ce639)

I found that this firmware is not usable, its speed is much slower than the old version on mbed, the current and control loop can not run to 40K.