MiRoboticsLab/cyberdog_ros2 #112: 按照文档的编译环境，没有一个能跑通

whlsxl, 2021-12-02:
按照文档操作，从docker 到ubuntu 20.04 的虚拟机，编译都报错，丢包少库。

连docker 都少库，这个文档有检验过吗？

https://github.com/MiRoboticsLab/cyberdog_ros2/wiki/%E4%BB%8E%E6%BA%90%E7%A0%81%E5%AE%89%E8%A3%85ROS-2-%7C-Building-ROS-2-from-source

homalozoa, 2021-12-02:
有问题贴log。

whlsxl, 2021-12-02:
```
Finished <<< osrf_pycommon [3.80s]
Starting >>> ament_cmake_core
--- stderr: iceoryx_utils
In file included from /tmp/build/src/eclipse-iceoryx/iceoryx/iceoryx_utils/include/iceoryx_utils/internal/posix_wrapper/access_control.hpp:22:0,
                 from /tmp/build/src/eclipse-iceoryx/iceoryx/iceoryx_utils/source/posix_wrapper/access_control.cpp:17:
/tmp/build/src/eclipse-iceoryx/iceoryx/iceoryx_utils/platform/linux/include/iceoryx_utils/platform/acl.hpp:19:10: fatal error: sys/acl.h: No such file or directory
 #include <sys/acl.h>
          ^~~~~~~~~~~
```
这是docker build 之后编译的错误，
`sudo apt-get install uuid-dev libacl1-dev liblzo2-dev -y`解决之后
```
--- stderr: tinyxml2_vendor
CMake Error at /usr/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find TinyXML2 (missing: TINYXML2_LIBRARY TINYXML2_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindTinyXML2.cmake:58 (find_package_handle_standard_args)
  CMakeLists.txt:7 (find_package)
```

在ubuntu 20.04 上安装依赖
```
The following packages have unmet dependencies:
 python3-colcon-ros : Depends: python3-catkin-pkg-modules (>= 0.4.14) but it is not going to be installed
 python3-rosdep-modules : Depends: python3-rospkg-modules (>= 1.3.0) but it is not going to be installed
                          Depends: python3-catkin-pkg-modules (>= 0.4.0) but it is not going to be installed
                          Depends: python3-rosdistro-modules (>= 0.7.5) but it is not going to be installed
```

在cyberdog 上直接编译 https://github.com/MiRoboticsLab/cyberdog_ros2/issues/55 同样的错误。

homalozoa, 2021-12-02:
Hi@whlsxl ，你确定按照文档里一步步安装依赖了吗？另外，20.04的AMD64docker并不是用来直接编译ROS 2的，那个是还没配置完毕的交叉编译环境，想直接编译ROS 2本体（不含cyberdog_ros2），可以用arm64的docker，跑在qemu里的。

whlsxl, 2021-12-02:
想用docker，可以不用污染server 的环境。如果20.04可以跑通的话，那docker 环境应该也可以跑通啊。

homalozoa, 2021-12-02:
可以用AMD64 Ubuntu 20.04的docker跑一下那个教程文档，有问题随时在本issue提出来。

另外，源码编译分为两部分，一个是ROS 2的本体，一个是本项目。二者的问题请区分开讲，不然我也分不清你说的是哪个，也不好答复你。

whlsxl, 2021-12-02:
我的目的肯定是要编译cyberdog，我也是第一次接触ros2，只能按照文档跑，现在按照文档还没跑起来。

whlsxl, 2021-12-02:
> 可以用AMD64 Ubuntu 20.04的docker跑一下那个教程文档，有问题随时在本issue提出来。
> 
> 另外，源码编译分为两部分，一个是ROS 2的本体，一个是本项目。二者的问题请区分开讲，不然我也分不清你说的是哪个，也不好答复你。

文档里没有关于docker 的文档。

你的意思是，在docker里，手动安装依赖吗？

whlsxl, 2021-12-02:
这个文档里的是空的。

https://github.com/MiRoboticsLab/cyberdog_ros2/wiki/%E4%BD%BF%E7%94%A8Docker%E6%9E%84%E5%BB%BA%E9%93%81%E8%9B%8B%E9%A1%B9%E7%9B%AE-%7C-Building-CyberDog-Projects-with-Docker

homalozoa, 2021-12-02:
> > 可以用AMD64 Ubuntu 20.04的docker跑一下那个教程文档，有问题随时在本issue提出来。
> > 另外，源码编译分为两部分，一个是ROS 2的本体，一个是本项目。二者的问题请区分开讲，不然我也分不清你说的是哪个，也不好答复你。
> 
> 文档里没有关于docker 的文档。
> 
> 你的意思是，在docker里，手动安装依赖吗？

如果是[arm64的docker](https://github.com/MiRoboticsLab/cyberdog_ros2/blob/devel/tools/dockers/arm64/Dockerfile)，Dockerfile里有完整的构建流程（包括从源码构建ROS 2），只不过由于QEMU效率的问题，会很耗时。当然，如果你有Apple的M1、M1 Pro或M1 Max会极大程度上提高这一构建效率。

如果是[amd64的docker](https://github.com/MiRoboticsLab/cyberdog_ros2/blob/devel/tools/dockers/amd64/Dockerfile)，不推荐你在这里编译ROS 2，因为二者架构不同，你的目的是编译本项目并部署至铁蛋，目前无法满足这个需求。

homalozoa, 2021-12-02:
> 这个文档里的是空的。
> 
> [使用Docker构建铁蛋项目 | Building CyberDog Projects with Docker ](https://github.com/MiRoboticsLab/cyberdog_ros2/wiki/%E4%BD%BF%E7%94%A8Docker%E6%9E%84%E5%BB%BA%E9%93%81%E8%9B%8B%E9%A1%B9%E7%9B%AE-%7C-Building-CyberDog-Projects-with-Docker)

> Hi@whlsxl ，你确定按照文档里一步步安装依赖了吗？另外，20.04的AMD64docker并不是用来直接编译ROS 2的，那个是还没配置完毕的交叉编译环境，想直接编译ROS 2本体（不含cyberdog_ros2），可以用arm64的docker，跑在qemu里的。

这里已经提到了，目前不支持交叉编译，也不支持在Docker里构建后部署，需要后续逐步做支持。

whlsxl, 2021-12-02:
在m1 跑qumu的 ubuntu 20.04，用 `docker build -t arm_docker:1.0 . --build-arg gfw=1` 编译报错。

```
--- stderr: fastrtps
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [src/cpp/CMakeFiles/fastrtps.dir/rtps/writer/RTPSWriter.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [src/cpp/CMakeFiles/fastrtps.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< fastrtps [3min 49s, exited with code 2]
```

homalozoa, 2021-12-02:
> 

内存炸了，分配大一点

whlsxl, 2021-12-03:
```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
flake8 4.0.1 requires importlib-metadata<4.3; python_version < "3.8", but you have importlib-metadata 4.8.2 which is incompatible.
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
```
这是build docker arm image 时候的报错

```
--- stderr: iceoryx_posh
Cloning into 'src'...
HEAD is now at fededad one-char typo
Submodule 'deps/meta-cmake' (https://github.com/meta-toolkit/meta-cmake.git) registered for path 'deps/meta-cmake'
Cloning into '/root/build_ros/build/iceoryx_posh/dependencies/cpptoml/src/deps/meta-cmake'...
fatal: unable to access 'https://github.com/meta-toolkit/meta-cmake.git/': gnutls_handshake() failed: The TLS connection was non-properly terminated.
fatal: clone of 'https://github.com/meta-toolkit/meta-cmake.git' into submodule path '/root/build_ros/build/iceoryx_posh/dependencies/cpptoml/src/deps/meta-cmake' failed
Failed to clone 'deps/meta-cmake'. Retry scheduled
Cloning into '/root/build_ros/build/iceoryx_posh/dependencies/cpptoml/src/deps/meta-cmake'...
fatal: unable to access 'https://github.com/meta-toolkit/meta-cmake.git/': gnutls_handshake() failed: The TLS connection was non-properly terminated.
fatal: clone of 'https://github.com/meta-toolkit/meta-cmake.git' into submodule path '/root/build_ros/build/iceoryx_posh/dependencies/cpptoml/src/deps/meta-cmake' failed
Failed to clone 'deps/meta-cmake' a second time, aborting
CMake Error at download/ext_cpptoml-prefix/tmp/ext_cpptoml-gitclone.cmake:52 (message):
  Failed to update submodules in:
  '/root/build_ros/build/iceoryx_posh/dependencies/cpptoml/src'

make[2]: *** [ext_cpptoml-prefix/src/ext_cpptoml-stamp/ext_cpptoml-download] Error 1
make[1]: *** [CMakeFiles/ext_cpptoml.dir/all] Error 2
make: *** [all] Error 2
CMake Error at cmake/cpptoml/CMakeLists.txt:61 (message):
  Build step [download] for cpptoml failed: 2

---
Failed   <<< iceoryx_posh [4min 3s, exited with code 1]
```
非常难编译，即使已经翻墙了，但是经常还是会卡在这种问题上面。

```
CMake Error at /opt/ros2/galactic/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:77 (message):
  ament_target_dependencies() the passed package name 'OpenCV' was not found
  before
Call Stack (most recent call first):
  CMakeLists.txt:157 (ament_target_dependencies)
```
`docker build`了两次，都有这个错误。`pip install opencv-python` 和 `apt install python3-opencv` 都没效

### 建议：

docker 可以push 到[docker hub](https://hub.docker.com/)上，这样省去很多时间。

其实我的需求很简单，只是想看看最新的固件到什么程度，调用下grpc 看看能做什么，但是找不到地方下载一个`nightly build`。

homalozoa, 2021-12-03:
没经费用在docker hub上，不好意思。网络问题的话，建议仔细阅读readme，里面有应对的策略。

目前devel分支在大幅度改动，会有很多功能不可用/不好用的情况，不熟悉ROS 2或者铁蛋功能的情况下建议不要部署到机器上，否则只能刷机恢复原厂。

whlsxl, 2021-12-03:
docker hub 开放的image 是免费的，只是用他的自动编译收费。

whlsxl, 2021-12-03:
请问能不能有个安装文件的包，直接安装在cyberdog 上，就能恢复cyberdog，而不是每次都要刷机。

homalozoa, 2021-12-03:
> docker hub 开放的image 是免费的，只是用他的自动编译收费。

免费账户拉取次数是有限的。

whlsxl, 2021-12-03:
谢谢！

whlsxl, 2021-12-03:
还有个问题，现在不连接app， app_server 没有打开，不能grpc通讯。
可不可以设置grpc 常开？

closed 2021-12-03 by whlsxl

homalozoa, 2021-12-03:
> 还有个问题，现在不连接app， app_server 没有打开，不能grpc通讯。
> 可不可以设置grpc 常开？

这个我们在内部实现了一个快速连接的版本，最近我在推动适配到开源版本。

whlsxl, 2021-12-03:
请问现在有能直接启动grpc的方法么

homalozoa, 2021-12-03:
可以直接发送String类型的Topic启动，参考https://github.com/MiRoboticsLab/cyberdog_ros2/blob/devel/cyberdog_common/cyberdog_grpc/src/cyberdog_app.cpp#L445

whlsxl, 2021-12-03:
`ros2 topic pub /mi1035949/ip_notify std_msgs/msg/String "data: [ip]"`

这样可以启动了

wu1999a, 2021-12-26:
> `ros2 topic pub /mi1035949/ip_notify std_msgs/msg/String "data: [ip]"`
> 
> 
> 
> 这样可以启动了

你好 想请教一下怎么能启动grpc来控制机械狗

katrina1874, 2022-11-18:
你好 想请教一下在win7上装Fast DDS安装过程，其中Fast-rtps构建失败，报错是"CMake Error: cannot find package TINYXML2"，我检查tinyxml2下的include目录后发现确实只有tinyxml2.h和另一个git文件，但之前在好几台电脑上试过，没问题，不知道怎么在这台上就报错了...亟待解决，不胜感激