MiRoboticsLab/cyberdog_ros2 #113: colcon build error

bigbencat, 2021-12-05:
1. git pull
2.build:
 `mi@lubuntu:~/ros_apps$ colcon build --merge-install --packages-up-to cyberdog_bringup
Starting >>> cascade_lifecycle_msgs
Starting >>> ception_msgs                             
Starting >>> interaction_msgs
Starting >>> lcm_vendor                                                                                           
Starting >>> motion_msgs
Starting >>> toml11_vendor                                                                                                                                 
--- stderr: lcm_vendor                                                                                                                                      
CMake Error: The source directory "/home/mi/ros_apps/src/cyberdog_ros2/cyberdog_common/lcm_vendor" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
make: *** [cmake_check_build_system] Error 1
---
Failed   <<< lcm_vendor [0.98s, exited with code 2]
--- stderr: toml11_vendor                                                                                                                                  
CMake Error: The source directory "/home/mi/ros_apps/src/cyberdog_ros2/cyberdog_common/toml11_vendor" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
make: *** [cmake_check_build_system] Error 1
---
Failed   <<< toml11_vendor [1.19s, exited with code 2]
Aborted  <<< cascade_lifecycle_msgs [2.96s]                                                                                                                
Aborted  <<< ception_msgs [3.24s]                                                                                                               
Aborted  <<< motion_msgs [3.12s]                                                                                 
Aborted  <<< interaction_msgs [3.54s]                                 

Summary: 0 packages finished [7.82s]
  2 packages failed: lcm_vendor toml11_vendor
  4 packages aborted: cascade_lifecycle_msgs ception_msgs interaction_msgs motion_msgs
  2 packages had stderr output: lcm_vendor toml11_vendor
  8 packages not processed
`

bigbencat, 2021-12-05:
display error when open terminal window:
`not found: "/opt/ros2/cyberdog/share/behaviortreecppv3_vendor/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/ception_base/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/cyberdog_bt_engine/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/cyberdog_ception_bridge/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/bcmgps_plugin/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/cyberdog_utils/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/cyberdog_decisionutils/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/cyberdog_gps/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/cyberdog_lightsensor/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/cyberdog_motion_bridge/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/cyberdog_obstacle_detection/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/lcm_vendor/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/cyberdog_bms/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/cyberdog_body_state/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/cyberdog_interfaces/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/cyberdog_grpc/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/cyberdog_led/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/cyberdog_scene_detection/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/cyberdog_touch/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/mpg123_vendor/local_setup.bash"
not found: "/opt/ros2/cyberdog/share/sdl2_vendor/local_setup.bash"
mi@lubuntu:~$ 
mi@lubuntu:~$ 

`

homalozoa, 2021-12-05:
Remove 'build' dir , please. And try again.

closed 2021-12-12 by homalozoa