MiRoboticsLab/cyberdog_ros2 #114: ROS2 control of Cyberdog

tomek-l, 2021-12-16:
Hi! 

I'm using the robot and so far I'm really impressed with it, awesome work!
I was able to get [sample gRPC](https://github.com/Karlsx/CyberDog_Ctrl) code running, and build some sample apps on top of that.

I'm curious whether there are any simple examples of operating using ROS2 directly?  If not, do you have any recommendations on how to approach it? So far, the best idea I have is to instantiate `Cyberdog_app` Node and go from there.

Best,
Tomasz

homalozoa, 2021-12-16:
Hi Tomasz. Thanks for your attention! There are many ways to control cyberdog, like using gRPC, ROS 2 messages, LCM, etc. We use gRPC as a very high level to control the robot, because it support so many programming languages. But ROS messages is only support C++ and Python, if you do not install third party RCL implementation. LCM is only designed for locomotion controling, we use it because MIT mini cheetah using it. The locomotion is built on their exellent work.

You can do more things by using ROS messages, but I'm refactoring the whole project recently. Hince many APIs will change in Future. APIs are including all controlling cmds, sensor datas, output devices cmds.

But if you want to control robot only. Please refer to https://github.com/MiRoboticsLab/cyberdog_ros2/tree/devel/cyberdog_interfaces/motion_msgs and https://github.com/MiRoboticsLab/cyberdog_ros2/blob/devel/cyberdog_managers/README_EN.md . I forgot writing English README for interfaces, you could translate it with Google Translation. If you have any problem, please let me know. And post in this issue. Thank you.

homalozoa, 2021-12-21:
By the way. As we chatted at Telegram. Publishing the IP of the HOST device as a string to topic "ip_notify" is the best way to active cyberdog with gRPC. The HOST device could be your mobile phone, or your PC in the same local area network or wireless local area network.

closed 2021-12-21 by homalozoa

hno3kyoz, 2022-01-20:
https://github.com/MiRoboticsLab/cyberdog_ros2/blob/devel/cyberdog_managers/README_EN.md

hno3kyoz, 2022-01-20:
The url:
https://github.com/MiRoboticsLab/cyberdog_ros2/blob/devel/cyberdog_managers/README_EN.md 
was changed to 
https://github.com/MiRoboticsLab/cyberdog_ros2/blob/devel/cyberdog_decision/README_EN.md