MiRoboticsLab/cyberdog_ros2 #121: Cyberdog在运动控制解算系统中是否使用实时内核

mailonghua, 2021-12-23:
运动结算放在linux中进行，而linux属于非实时的系统，所以cyber中是否对linux打入实时内核？什么内核？
运动控制指令的发送频率是多少呢

homalozoa, 2021-12-23:
### 关于实时内核
由于运动控制程序是基于[MIT ](https://github.com/mit-biomimetics/Cheetah-Software)修改的，所以在另外一块主板上，是基于全志的MR813，我们在它的Linux内核上面增加了Preempt RT Patch，由于这部分的CI没有整理好，所以仅开放在小米的服务器上：[athena_tina_sdk](https://partner-gitlab.mioffice.cn/cyberdog/athena_tina_sdk)

### 关于运动指令的频率

#### 如果你问的是从Xavier NX --> MR813
这个频率是可以修改的，在[rate_control_hz](https://github.com/MiRoboticsLab/cyberdog_ros2/blob/devel/cyberdog_bringup/params/default_param.yaml#L41)，单位是频率，但不能超过500。

#### 如果你问的是运控程序对电机的发送频率
如上所述的最高限制：500hz。

mailonghua, 2021-12-23:
> ### 关于实时内核
>
> 由于运动控制程序是基于[MIT ](https://github.com/mit-biomimetics/Cheetah-Software)修改的，所以在另外一块主板上，是基于全志的MR813，我们在它的Linux内核上面增加了Preempt RT [...]

多谢解答，目前Cyber的控制算法均开源了吗？在上面的提到的链接中

homalozoa, 2021-12-25:
> 多谢解答，目前Cyber的控制算法均开源了吗？在上面的提到的链接中
 
可以翻阅其他issue，或discussion。这部分算法程序不会开源，但会以SDK形式释放。

closed 2021-12-25 by homalozoa