MiRoboticsLab/cyberdog_ros2 #128: PREEMPT_RT做了哪些优化，如何确保指令下发定时器的延迟抖动

mailonghua, 2022-01-21:
PREEMPT_RT做了哪些优化，如何确保指令下发定时器的延迟抖动

Qiupetter, 2022-01-22:
你好，铁蛋运动控制系统内核采用开源的Linux RT patch来增强系统实时性（CONFIG_PREEMPT_RT_FULL），主要优化点在中断线程化，自旋锁互斥化，以及优先级翻转等方面。

patch源码：https://www.kernel.org/pub/linux/kernel/projects/rt

    Fully Preemptible Kernel (RT): 实时系统

All kernel code is preemptible except for a few selected critical sections. Threaded interrupt handlers are forced. Furthermore several substitution mechanisms like sleeping spinlocks and rt_mutex are implemented to reduce preemption disabled sections. Additionally, large preemption disabled sections are substituted by separate locking constructs. This preemption model has to be selected in order to obtain real-time behavior.

mailonghua, 2022-01-24:
> 你好，铁蛋运动控制系统内核采用开源的Linux RT patch来增强系统实时性（CONFIG_PREEMPT_RT_FULL），主要优化点在中断线程化，自旋锁互斥化，以及修正优先级翻转等方面。
> 
> 
> 
> patch源码：https://www.kernel.org/pub/linux/kernel/projects/rt
> 
> 
> 
>     Fully Preemptible Kernel (RT): 实时系统
> 
> 
> 
> All kernel code is preemptible except for a few selected critical sections. Threaded interrupt handlers are forced. Furthermore several substitution mechanisms like sleeping spinlocks and rt_mutex are implemented to reduce preemption disabled sections. Additionally, large preemption disabled sections are substituted by separate locking constructs. This preemption model has to be selected in order to obtain real-time behavior.
> 
> 

你好，单纯的使用PREEMPT_RT应该不能完全解决一些问题，比如cyberdog每2ms发送一次指令，那么如何保证2ms精度

mailonghua, 2022-02-14:
有伙伴帮忙解惑一下吗

mi-CyberDog, 2022-02-14:
软实时操作系统并不能完全确保2ms的精度

mailonghua, 2022-02-15:
> 软实时操作系统并不能完全确保2ms的精度

但过大的延迟抖动会影响在高速条件下，机器狗的运动状态
在cyberdog中有测试过系统+链路的总延迟是多少吗

mailonghua, 2022-02-19:
有伙伴帮忙解答吗