MiRoboticsLab/cyberdog_ros2 #163: CyberDogAPP Where is the source code repository?

vitorcalvi, 2023-01-28:
## Brief modification description
I didn't find the CyberDogAPP source code to compile and make some own version

MAVProxyUser, 2023-01-31:
manager of the team refuses to give it out. 

sadly killing his own product

vitorcalvi, 2023-01-31:
Thank you for the reply.
So it's not opensource project

MAVProxyUser, 2023-01-31:
@vitorcalvi sadly they sold us on this, but are not honoring the commitment to open source. 
I understand the manager xiangdiyun@xiaomi.com does not care. Email him and ask why? Maybe if enough of us mail him he will care? 

Notice how we have no Gazebo support either? It is a joke.

vitorcalvi, 2023-01-31:
We have to join and kindly ask xiangdiyun@xiaomi.com to really open the source code to who bought the cyberdog as an opensource product.
I will send now kindlying ask about it.
Let's do the same too